mikeobrien/HidLibrary #52: Multiple same type devices, device path included in OnReport?

jdahlblom, 2015-02-25:
I am using this excellent library for the Saitek DCS Flightpanels project.
How can I differentiate HID input reports from same type devices?
I tried to see if it was possible to inject the unique Windows Device Path into the callback call for OnReport but my knowledge on WinAPI is close to zero and Google, Pinvoke didn't find enough info for me to work on. As OnReport callback must be Static and no uniqueness are included in the final callback call I don't know which device sent it.
Am I missing something? Thanks for this library, it is a gem.

amullins83, 2015-02-25:
I'd love to help with this issue if I can. I'm having trouble picturing the situation. If you have an `HidDevice` object, you should be able to call `device.ReadReport(callback)`, and that will call the given function when that specific device delivers its report. The callback passed to `ReadReport` does not have to be static.

jdahlblom, 2015-02-26:
Thank you! You are absolutely right. I have no idea why I got stuck on this. Well I am alone so there are a lot of paths to get stuck on. Oh man so much you helped by just writing your comment. I had to go through my code and surely it works.
THANKS!

closed 2015-02-26 by jdahlblom