mikeobrien/HidLibrary #53: WriteReportAsync - await Task

Pukkai, 2015-06-15:
Hi!

I'm beginner, so maybe I alone to getting this issue :) 

Problem is, when I want to send something with WriteReportAsync (or also WriteReport), report Is NOT send immediately, but is registered when I push button on my device. With any other USB sending program (like "SimpleHIDwrite") works normally. Only now with this C# library I get this issue, always when I want to send something to device, I must press button on my device.

My code stops here (HidDevice.cs):
return await Task<bool>.Factory.FromAsync(writeReportDelegate.BeginInvoke, writeReportDelegate.EndInvoke, report, timeout, null);

.. waiting for button on my device to be pressed.

Please for any help!

amullins83, 2015-06-15:
In my experience, this type of issue is usually caused by the device itself. If you have it, look at the source code for SimpleHIDwrite and see what that library does differently at the system call level. Things that come to my mind are buffer size inconsistencies, control vs interrupt endpoint confusion, and controlling the packet byte structure. I'm not familiar with SimpleHIDwrite, but those are the kinds of things I would look at. If you control or have access to the device firmware, try to see what happens on a button press that might allow the USB I/O to "wake up" if it has gone into a low-power state.

Pukkai, 2015-06-18:
OK. I dont know if is correct or not, but by the luck I add to code before sending "HidDevice.OpenDevice()" and after I send "HidDevice.CloseDevice();" I did not have time yet to overview SimpleHIDwrite. Sorry.

And my next question will be, how to read device info (ReadManufacturer, ReadSerialNumber, ReadProduct). I overview your code (HIDlibrary) very fast and how much I see it is already implemented with return method. You have any example how to read it? Because I am beginner in C# :) THANKS for fast reponse and all the help!