mikeobrien/HidLibrary #54: Timeout in ReadReport method does not work

gnatyukv, 2015-07-12:
mikeobrien / HidLibrary / src / HidLibrary / HidDevice.cs

In order to read the port, I am using the ReadReport method. But when a command did not came and no data is available, this method became unresponsive. When I am trying to set a timeout into an overloaded version of this method (ReadReport(1000), for example), this does not work and became unresponsive too. So, the timeout in your library does not work.
The same things is related to the ReadReportAsync method and, likely, to the writing methods as well.

Barmallini, 2015-07-12:
I had same problem. Two days and nights :)
Solution:
_device.OpenDevice(DeviceMode.Overlapped,DeviceMode.Overlapped,ShareMode.Exclusive);

But still have another problem - lost reports from device.
Read event fire, but report buffer empty (ReportID = 0, and rest of buffer = 0) with status "Success"

Pukkai, 2015-08-11:
Same problem here.
I have no luck with your solution Barmalllnl.

When I choose device (from ListView), after that I open device. When debugging I see thats open like I set up (you line of code), but TimeOut still not working

var _choosendevice = HidDevices.Enumerate(ChoosenDevice.DevicePath).FirstOrDefault();
_choosendevice.OpenDevice(DeviceMode.Overlapped, DeviceMode.Overlapped, ShareMode.Exclusive);

Any suggestion? :)

Pukkai, 2015-10-01:
Sorry Barmalllni ... your solution works, I have already OpenDevice by default ;) So CloseDevice and reopen with your line helps. THANKS!

amullins83, 2016-06-28:
I'm starting some issue cleanup (since I have commit privileges now). Is this an actual issue, or does everything work well when the device is opened with the correct overlapped and sharing modes?

gnatyukv, 2016-06-29:
amullins83, you are right. I had to see a code flow before asking a question, and timeout does not affect. Timeout acts only in overlapped mode. So my message "Timeout in ReadReport method does not work" has no sense. But in overlapped mode another problem is occurred. Zero responds are coming. But actually, I think, this low-level timeout is not needed. When no respond from the HID device, the following steps may be processed:

1. CloseDevice();
2. OpenDevice();
3. After that zero respond is coming, should be ignored. On this stage, it seems as an unmanaged resources are released.
4. And now, HidLibrary is ready to normal work.

Otherwise, HidLibrary will not work, because its execution stops in the HID driver libraries, which are an unmanaged code. I mean, if you send a command and no respond back. And any tricks (such as sending each command in a new thread) will not help.

amullins83, 2016-06-29:
Thanks, @gnatyukv. That sounds similar to what @jvdub22 is describing in issue #67.

jvdub22, 2016-06-29:
so issue #54 looks to be the same issue I'm experiencing, do you still think it is on the device end?

thanks.

aivarsi, 2016-10-08:
@Barmallini Did you manage to solve the issue with losing reports? I have exactly the same problem - sometimes I get buffer that is filled with zeroes whether I use Read(Callback) or ReadReport(Callback). 

I'm using the NuGet version. 

I know it's not the device sending empty buffers because when I monitor the incoming data with another software I can see that the correct reports were always sent.

frankschierle, 2017-05-11:
I recently had the same problem. The device I'm working with is confirming the execution of every command it receives. Some commands need a few milliseconds, some of them need more than a second to execute. Lots of reports were lost while waiting for the confirmation of the long running commands. So I did some investigation and I think i found the problem causing the reports to be lost:

`HidDevice.ReadData(int)` creates a new auto-reset event and a new buffer for every read attempt. The problem is: After a timeout occurs, the asynchronous `ReadFile()` operation is not aborted. It is still running in the background. If a report from the device arrives between the timeout of a first read attempt and the start of a second read attempt, the report is received by the first call of `ReadFile()` and gets discarded (because the buffer it is written to is no longer in use). The second call to `ReadFile()` (made by a second read attempt) will not receive the report because it uses a new (empty) buffer.

[image]

In my application, I temporarely increased the timeout for read operations to avoid losing reports.