mikeobrien/HidLibrary #59: Magnetic card swipe Windows 10

paschka76, 2016-03-01:
A strange behavior for the MagtekCardReader example can be reproduced on Windows 10. 
If the reader is connected before Windows is started, the device is found, is successfully opened, all events (Attached, Removed, OnReport) are firing, but, when a card is swiped, in "OnReport" the IsConnected property is always false and no data is available in the report. Removing and attaching the reader helps - all subsequent swipes are fine and the report data can be read.
I can reproduce it on 2 machines in the office and a many clients with Windows 10 also reported this behavior.
Funny thing, if I restart the computer, everything works as expected. The problem appears only when the computer is shut down and the started.

I'm using the latest release (3.2.31)
Did anyone have same problems?
Thanks!

@mikeobrien, thank you for the library, especially for the MagtekCardReader example!

mikeobrien, 2016-03-01:
Haven't run into that before. I wrote this for a project back in the XP days so wouldn't be surprised if there are issues on new OS's, there have been a few things that have cropped up over the years. 

No problem, hope it's helpful! Sorry I haven't been able to support it much these days. There have been some active users that may have run into this.

paschka76, 2016-03-01:
@mikeobrien, can you please tell me what ReadReport does? Why is it needed after opening the device, attaching it and, especially, after handling the OnReport?

amullins83, 2016-03-01:
@paschka76 `ReadReport` reads one input report from the device and calls the given callback function on completion or error. 

In the example code, `OnReport` is the name of a callback function passed to `ReadReport`. It happens to call `ReadReport` with itself as the argument after processing the report so that the program is constantly retrieving new data from the device as it becomes available. This is not *needed* by any means, but it's a common pattern for device communication.

paschka76, 2016-03-01:
I see. So, ReadReport will _start waiting_ for the next, in my case, card swipe event? And after handling the event, I need to _start this waiting_ yet again?

amullins83, 2016-03-01:
Yes. HIDs have to be driven by the host, so any application code has to explicitly request data from the device. I can imagine a library being implemented in such a way that you could just attach an event handler to a .NET event and it would internally run a read loop. This is not that library.

paschka76, 2016-03-07:
@amullins83, thanks for the explanation

Meanwhile I stumbled upon a PC with Windows 10 where this issue can not be reproduced. So it's not a general Windows 10 problem, but I also wouldn't say that it's some issue with a specific PC, since there are at least 4 different PCs on which the bug can be reproduced. 

Any clues anyone where I could _dig_ to get closer to the problem?

Thanks!

paschka76, 2016-03-08:
What happens when I call ReadReport, let's say, twice? Does the later cancel the initial, or will it _wait_ for 2 reports?

amullins83, 2016-03-08:
It will wait twice. There is no attempt made to cancel a current read operation.

amullins83, 2016-06-29:
@paschka76 Microsoft has released a few updates to Windows 10 since our last communication. Have you still noticed this issue on up to date systems?