mikeobrien/HidLibrary #62: ReadProduct fails to return data

63degrees, 2016-03-15:
The underlying code in NativeMethod.cs HidD_GetProductString needs a larger buffer.
It is set at 64 bytes. MS API states 126 wide (see below). A wide character is a 2-byte multilingual character code.  I changed the ReadProcduct byte array to 252 and it worked for me.

https://msdn.microsoft.com/en-us/library/ff539681.aspx

The maximum possible number of characters in an embedded string is device specific. For USB devices, the maximum string length is **126 wide characters** (not including the terminating NULL character).

If the buffer is not large enough to return the entire NULL-terminated embedded string, the routine returns nothing in the buffer.

amullins83, 2016-03-15:
Submit a pull request with your fix. I would make the buffer 254 bytes wide so there is space for the terminating null.

63degrees, 2016-03-15:
Agree with the byte buffer 254 length.

commit 6357383a (2016-03-16): Merge pull request #64 from amullins83/master

Fix #62: Increase buffer size for device string read operations.
commit d28a7ba5 (2016-03-16): Fix #62: Increase buffer size for device string read operations.
closed 2016-03-16 by mikeobrien