mikeobrien/HidLibrary #63: Issue with ReadReport method with new lib version

lukeworld, 2016-03-16:
Hi, I always used this hid library in my .NET projects (4.0 and above) ver. I wanted to upgrade my lib version, I use the HidLibrary-3.2.28.0, to the newest HidLibrary-3.2.32.0 published few days ago, but I've noticed that my projects doesn't work anymore. Simply updating library and rebuilding solution, when the Write method is called, the sw hangs on this method and the OnReport callback is not being called. I tried to replace library with my old version and all come back to works. I tried even the version 3.2.29.0 but it reproduce the same issue.
My code is implemented as suggested in [MagTek CardReader](https://github.com/mikeobrien/HidLibrary/blob/master/examples/MagtekCardReader/Program.cs) using ReadReport and OnReport Callback.

Can someone help me to understand why my code doesn't work anymore with new version?
Thanks to all.

lukeworld, 2016-03-21:
Noone has issues replacing with new lib version......?

mikeobrien, 2016-03-21:
I haven't used the library in years so not sure what could be going on but I'd suggest posting your code.

amullins83, 2016-03-21:
Which versions of Windows show this problem with your application? Version 3.2.29 introduced a dependency on .NET 4.5 with the new Async/Await task style. .NET 4.5 can't be installed on Windows XP or older, but it should be there by default on Windows 8 and above. The iffy case is Windows 7, which is compatible with .NET 4.5 but I don't think it's on there by default.

If you need backwards compatibility to Windows XP, you should stick with HidLibrary version 3.2.28. I apologize that we did not indicate a potentially breaking change by incrementing the major version number.

lukeworld, 2016-03-21:
I use library on Windows 7 WPF app with Framework 4.5.2.
I tried to create a new app for test it, but I have the same issue, with old lib version code is working, with newer version the code stuck on write method of lib.
Thanks all for your help.

amullins83, 2016-03-21:
Thanks for narrowing it down. Can you post the most relevant parts of your code here to help us figure this out? What device are you using exactly? The MagTek card reader, or something else?

lukeworld, 2016-03-22:
Thanks for your reply. I use this library with a custom USB device, to communicate with custom electronic devices. I have a main worker thread that waits for bytes stream to send to a connected device, with OnReport event I get the answer from device and then I process it to my UI. As I said in above post, I tried to make a test project only to send a simple write to my custom usb device, but sw get stuck on this statement. If i remove USB the sw get unstuck.
Could anyone tell me if with MagTek or other usb device there are no issues with new lib versions, because in this case, I focus on Usb device firmware even on SW.
Thanks

amullins83, 2016-03-22:
First off, I was totally wrong about the problem having anything to do with .NET 4.5. It looks like the biggest change to the library was that, starting with 3.2.29, we're not opening separate read and write handles for each device. There's just one handle opened. Even though the `OpenDevice` function still takes a `writeMode` argument, only the `readMode` is actually used. In other words, your specified `writeMode` (`NonOverlapped` or `Overlapped`) is being ignored. I imagine this could cause problems depending on your application.

I'm going to take some more time going over the changes for 3.2.29 and try to come up with some more possibilities, but without your custom device, I won't be able to verify anything for you. The first thing you should try is changing the `readMode` passed to `OpenDevice`. The default is `DeviceMode.NonOverlapped`, so try seeing if opening it with `DeviceMode.Overlapped` solves the issue.

lukeworld, 2016-03-29:
Sorry for my late reply, I used OpenDevice without specify any param, so I use NonOverlapped configuration (that I want). I tried to change this by passing read and write mode as Overlapped, and in this mode the sw doesn't stuck but only because the write method goes in timeout in async overlapped mode. I tried to set mixed mode of read/write config, but in this case the sw goes in exception and it being killed.

amullins83, 2016-03-29:
Thanks for trying. Looking over the diff of v3.2.28.0 and v3.2.29.0 again, I remain convinced that your problems began when we changed from having separate read and write handles to using a single handle. For most devices in most use cases, this works. I suspect that your application is attempting a synchronous read before a synchronous write. The read stalls because the device has no data to send, and the write times out waiting for the read to complete.

We could rule out this possibility if you could post the most relevant parts of your application code here.

lukeworld, 2016-03-29:
Thanks Amullins you are a magician.
I did a write to set a  specific setting to my custom USB device without doing a reading from usb because is not necessary, then I called the readReaport method to set the callback for the subsequent "usb messages", to manage the reading from usb.
I tried to remove the first write, and all it works even with the 3.2.33.0 lib version.
For now I've tested it with `OpenDevice(DeviceMode.NonOverlapped, DeviceMode.NonOverlapped, ShareMode.ShareRead Or ShareMode.ShareWrite)`

amullins83, 2016-03-29:
LOL, sounds like I had it exactly backwards but you worked it out anyway. You're the magician now.