mikeobrien/HidLibrary #65: Sometimes library lost data from device.

winstondiz, 2016-03-22:
Hello. 

I have some problems with library, when my device return answer from commands, sometimes library receive zeros byte array, but in sniffer I get valid information. Can you explain this

winstondiz, 2016-03-22:
For receive information I use ReadReport with callback. For send information I used WriteReportAsync. 

Best regards, Alexander

winstondiz, 2016-03-23:
But when i change some method's in **ReadData** i get stable work.
I change - **NativeMethods.ReadFile** method  to **FileStream.Read**(C# native method).
Information recived correctly and without distortion.

amullins83, 2016-03-23:
I'm sorry it's taking so long to get back to you on this. How exactly are you calling `FileStream.Read` in your version? This isn't expected to be more reliable than the native Win32 API because the `Handle` was opened with a native method, and native pointers won't move around during garbage collection cycles.

I think there may be a race condition between the async / await style `WriteReportAsync` and the synchronous `ReadReport`. Constructing a `SafeFileHandle` .NET object that you can use to open a `FileStream` might be slowing down the runtime enough to avoid this race condition.

My question is why you're using the async write method with the synchronous read. I'm not as familiar with the differences in the async / await style code, so I would defer to @neilt6 for help with that aspect.

neilt6, 2016-03-25:
`ReadReportAsync()` and `WriteReportAsync()` just call the synchronous versions using `Task<T>.Factory.FromAsync()`, which presumably executes them on a background thread. So if you're mixing synchronous and asynchronous calls, you'll get race conditions galore. You might even have problems calling multiple asynchronous methods at the same time, I doubt there's any implicit reentrancy protection...

amullins83, 2016-06-28:
I'm closing this due to lack of response from the issue's opener. We should probably update documentation to indicate that we are not supporting mixing of the async and synchronous read/write methods.

closed 2016-06-28 by amullins83