mikeobrien/HidLibrary #66: Magellan 1100i barcode reader

jayshu01, 2016-04-21:
I downloaded your Hid Library project and in the examples folder I saw that there is a sample for Magtek card reader, currently I'm working on a similar product called Magellan 1100i barcode reader. So what I did was to copy the code from magtek card reader sample and changed the GetCardData method, but I'm not able to make it work, could you please help.

Magtek card reader sample
if (data != null && data.Length == 337)
 {
     if (data[0] == 1 || data[1] == 1 || data[2] == 1)
     {
         Error = true;
         _errorMessage = "Error reading data";
         return null;
     }
     var cardData = new byte[data[3] + data[4] + data[5]];
     Array.Copy(data, 7, cardData, 0, data[3]);
     Array.Copy(data, 117, cardData, data[3], data[4]);
     Array.Copy(data, 227, cardData, data[3] + data[4], data[5]);
     return cardData;
 }

Datalogic Magellan 1100i
ProductId: 0x2602
VendorId: 0x05F9
The changed code, since the magellan 1100i has the data length of 8

if (data != null && data.Length == 8)
{
    if (data[0] == 1 || data[1] == 1 || data[2] == 1)
    {
        Error = true;
        _errorMessage = "Error reading data";
        return null;
    }
    var cardData = new byte[data[2] + data[3] + data[4]];
    Array.Copy(data, 1, cardData, 1, data[3]);
    Array.Copy(data, 2, cardData, data[2], data[3]);
    Array.Copy(data, 3, cardData, data[3] + data[4], data[5]);
    return cardData;
}

Thanks
Jay

amullins83, 2016-04-21:
First, if the data length is 8 bytes, why are you allocating the `cardData` array based on the `data` array? You only have 8 bytes, right? Just do:
```C#
var cardData =  new byte[8];
```

Similarly, these copy statements don't make any sense to me. If you actually had a variable amount of data, it would make sense to index into your destination based on the section lengths provided by the device. However, you know your data only has 8 bytes, so why not copy them all?

```C#
Array.copy(data, 0, cardData, 0, 8);
```

Really, it would be better to fill a custom structure that would make this data meaningful to your users, but that could be handled in another method. It looks like the goal of this method is to just get the data from the device into a buffer you own.

jayshu01, 2016-04-21:
Thanks Austin, by reading the full length it breaks on the callback method, OnReport,                 
if (cardData.CardData.Length > 0)
                {
                    Console.WriteLine(Encoding.ASCII.GetString(cardData.CardData));
                    SystemSounds.Asterisk.Play();
                } 

Since the length is always greater than 0, it keeps writing to the console.

amullins83, 2016-04-21:
What behavior would you prefer?

jayshu01, 2016-04-21:
So when I scan my barcode, I'm expecting a string to be returned from cardData.CardData, the format of the string is some what like this,

"POND DRDAISTOCKBRIDGEDAJGADAK30281-5192 DAQ055869536DAU064 inDAYBLUDBA11142009DBB03291960DBC2DBD09302009DCACMDCBADCDNONEDCF029600000370033759DCGUSADCHNONEDCSPERRYDCTLAURIE MITCHELDCE2
ZGZGANZGBRev06222009ZGCNZGENZGFNZGG120ZGH5-04ZGIHENRYZGK02960000017ZGL0043475709821ZGM038
|"

amullins83, 2016-04-21:
We're getting closer to the real issue now. That string is a lot longer than 8 characters. When you say the 1100i has a data length of 8, what does that mean really? Are you getting 8 characters of this string at a time? Do you need to write your application so that it concatenates these strings?

jayshu01, 2016-04-21:
So on the GetCardData(byte[] data) method when I see the length of the data variable, it shows me 8 (0-7), and when I scan my barcode, I see the long string, basically it's the driver license, and I need to parse the string to get the info.

amullins83, 2016-04-21:
Ok. My question is what are those bytes? You know that you get 8 of them at a time. Are they 8 characters from this string? Do some of the bytes represent other information?

Your application probably needs to build a string from multiple data reads since your data is coming in 8 bytes at a time and your high-level read event needs a longer string. So, in your handler for the HidDevice read report event, you need to append some data to your barcode result string, check to see if you have a full barcode, and if so, fire an event of your own to signal to the application that it's time to write out to the console (or parse the string to fill a custom data object, etc).

jayshu01, 2016-04-21:
I don't know what those bytes are, because they all have a value of 0, and when I scan, the string which I get is not stored anywhere, instead it writes wherever my cursor is positioned. Hope I'm clear, sorry.

amullins83, 2016-04-21:
Oh. That means Windows grabbed the barcode scanner as a keyboard. Once it does that, HidLibrary can't access it. Keyboards are a whole different API separate from HID. I'm not sure if there's a way to avoid that. It would probably involve installing a custom kernel-mode driver. HidLibrary only works in user-mode.

jayshu01, 2016-04-21:
Thanks Austin, appreciate your help. I will let you know if I resolve this issue.