mikeobrien/HidLibrary #79: ReadReport after Timeout

DevLocus, 2017-02-07:
ReadReport will always fail after a timeout happens the first time.  The only way I found to clear the issue is to close the handle and open a new one.  I tried CancelIOex, with the handle and a null pointer, when an error during ReadFile or timeout occurs.  But the ReadFile still times out while the analyzer shows data available.

ananth-racherla, 2017-02-25:
Can you please check if this is fixed in the latest?

teplofizik, 2017-05-25:
That helps: CancelIO if not Success:
```
case NativeMethods.WAIT_TIMEOUT:
                                    status = HidDeviceData.ReadStatus.WaitTimedOut;
                                    NativeMethods.CancelIo(Handle); // !!!
                                    buffer = new byte[] { };
                                    break;
```

leotang315, 2017-08-22:
thankyou , it work @teplofizik