mikeobrien/HidLibrary #85: Documentation

SoulMaril, 2017-05-22:
Is there any guide or documentation for how to integrate with HidLibrary. I am not able to find.

amullins83, 2017-05-22:
It's not clear what you need. The sample projects inside the repo itself are intended as a kind of documentation by way of example. Are you able to build the sample projects?

SoulMaril, 2017-05-23:
Yes. Currently I am using the examples. They are very helpful but I need a
shortcut.What I was asking was something like a reference guide.
I created some documents with doxygen and sandcastles for my self. As there
is no comments and summaries I have to figure out what is doing what by
using the examples.
I believe if I had a document this will speed up the time I spend for
trying to decipher the whole library.
Considering that I am not primarily a C# programmer, I am not that
confortable as most of you it will be very very practical for people like
me.

DiskCrasher, 2017-07-19:
I concur. There are no comments or XML documentation within the source code, which should be a requirement for any project designed to be used by others.