mikeobrien/HidLibrary #88: X56 Rhino flight stick issues

StainlessTobii, 2017-08-22:
Firstly, thanks for this library.

I had a horrible day of frustration when I discovered DirectInput was deprecated, XInput only does Xbox gamepads, and Game.Windows.Input is only for UWP.

Then I found this.

I found that the x56 throttle box worked perfectly with your code, but the joystick was very very slow.

The solution I used in the end was this.....

1) Use a fast device instead of a device.

```
            deviceList = HidDevices.Enumerate().ToArray();
            for (int i = 0; i < deviceList.Length; i++)
            {
                if (deviceList[i].Description.StartsWith("X56 H.O.T.A.S. Stick (HID)"))
                {
                    RhinoStick = deviceList[i];

                    HidFastReadEnumerator enumerator = new HidFastReadEnumerator();
                    fastStick = (HidFastReadDevice)enumerator.GetDevice(RhinoStick.DevicePath);

                    
                    fastStick.OpenDevice();
                    fastStick.MonitorDeviceEvents = false;
                    fastStick.Inserted += Device_Inserted;
                    fastStick.Removed += Device_Removed;
                    fastStick.FastReadReport(RhinoReadProcess);
```
2) Use an update thread instead of updating on the main thread

```
                    updateThread = new Thread(() => timer_elapsed(this));
                    updateThread.Start();
```

```
private static void timer_elapsed(X56HOTASStick parent)
        {
            while (parent.scan)
            {
                if (parent.fastStick != null)
                {
                    HidDeviceData dat = parent.fastStick.Read();
                    if (dat.Status == HidDeviceData.ReadStatus.Success)
                    {
                        HidReport report = new HidReport(parent.fastStick.Capabilities.InputReportByteLength, dat);
                        parent.RhinoReadProcess(report);
                    }
                }
            }
        }

```

Now I have no noticeable latency (it will be 2 frames, but I can't see that in game), and everything is cool.

Since this is hobby code and not work code, being able to use your library was a god send, I probably wouldn't have had the time to add joystick support to the game without it.

Cheers 

Paul

mikeobrien, 2017-09-26:
👍

evilC, 2018-10-14:
Thanks for this, I was implementing a reader for the SpaceMouse, got it all logically working,but it was slow.  
@StainlessTobii Check out the [UCR](https://github.com/Snoothy/UCR) project. We allow adding support for arbitrary devices to our Gui remapping app via C# plugins. See the Discord link in the repo if you wanna get in touch ;)