mikeobrien/HidLibrary #89: HidLibrary cannot write data on Windows 10

alidayan, 2017-09-20:
Hello everyone,
On my project I want to write data to PSoC device. I can connect and read datas but write. Here is my code:
```
private void sendData_Click(object sender, EventArgs e)
        {
            byte[] outData = new byte[_selectedDevice.Capabilities.OutputReportByteLength];
            outData[0] = Convert.ToByte(reportID.Text);
            outData[1] = Convert.ToByte(sData_1.Text);
            outData[2] = Convert.ToByte(sData_2.Text);
            outData[3] = Convert.ToByte(sData_3.Text);
            outData[4] = Convert.ToByte(sData_4.Text);
            outData[5] = Convert.ToByte(sData_5.Text);
            outData[6] = Convert.ToByte(sData_6.Text);
            outData[7] = Convert.ToByte(sData_7.Text);
            outData[8] = Convert.ToByte(sData_8.Text);

            _deviceList[Devices.SelectedIndex].Write(outData);
        }
```
First byte is report ID and follows by 8 bytes of datas. On windows 7 I can write data with UsbLibrary.dll. But I need Windows 10 support.

myn724405566, 2017-11-23:
I and you the opposite, I can write data to the device in WIN10 system, but WIN7 system can not write

easyvolts, 2018-05-04:
I have the same problem on Win10, WriteReport method isn't executed until some IN report from the device comes.

MelbourneDeveloper, 2018-07-08:
@alidayan  I don't think library will work in UWP. Can you confirm that your platform in UWP? Or, is it just normal .NET?

I could be wrong, but I tried using this library in UWP, and the problem I encountered was that I could not enumerate the devices at all. I think this is because Windows UWP does not allow you to use the normal windows APIs to connect to USB. I think you have to use the native UWP platform specific API for talking to USB.

I made this library : https://github.com/MelbourneDeveloper/Hid.Net . I use it on UWP. The important thing you have to remember in UWP is to define the DeviceCapability in the package manifest like this:

```
  <Capabilities>
    <Capability Name="internetClient" />
    <uap:Capability Name="removableStorage" />
    <DeviceCapability Name="humaninterfacedevice">
      <Device Id="vidpid:534C 0001">
        <Function Type="usage:0005 *" />
        <Function Type="usage:FF00 0001" />
        <Function Type="usage:ff00 *" />
      </Device>
    </DeviceCapability>
  </Capabilities>
```

vostrenkov, 2019-03-13:
In **NativeMethods.cs** changing `CharSet = CharSet.Auto` to `CharSet = CharSet.Unicode` for imported functions CreateFile and CreateEvent solved for me problem with WriteReport method on Windows 10 and DotNet Framework 4.6.2

Should be:
```
[DllImport("kernel32.dll", CharSet = CharSet.Unicode)]
static internal extern IntPtr CreateEvent(ref SECURITY_ATTRIBUTES securityAttributes, int bManualReset, int bInitialState, string lpName);

[DllImport("kernel32.dll", CharSet = CharSet.Unicode, SetLastError = true)]
static internal extern IntPtr CreateFile(string lpFileName, uint dwDesiredAccess, int dwShareMode, ref SECURITY_ATTRIBUTES lpSecurityAttributes, int dwCreationDisposition, int dwFlagsAndAttributes, int hTemplateFile);
```

vostrenkov, 2019-03-13:
I was able to solve problem with WriteReport on Windows 10 DotNet Framework 4.6.2 by setting different file handles for Read and Write operations:

**HidDevice.cs**
```
public IntPtr ReadHandle { get; private set; }
public IntPtr WriteHandle { get; private set; }

ReadHandle = OpenDeviceIO(_devicePath, readMode, NativeMethods.GENERIC_READ, shareMode);
WriteHandle = OpenDeviceIO(_devicePath, writeMode, NativeMethods.GENERIC_WRITE, shareMode);
```
**IHidDevice.cs**
```
IntPtr ReadHandle { get; }
IntPtr WriteHandle { get; }
```

instead of 

**HidDevice.cs**
```
public IntPtr Handle { get; private set; }

Handle = OpenDeviceIO(_devicePath, readMode, NativeMethods.GENERIC_READ | NativeMethods.GENERIC_WRITE, shareMode);
```
**IHidDevice.cs**
```
IntPtr Handle { get; }
```
and of course i had to set corresponding handle for every method where it needed, for example:

``` 
public HidReport ReadReportSync(byte reportId)
{
            byte[] cmdBuffer = new byte[Capabilities.InputReportByteLength];
            cmdBuffer[0] = reportId;
            // ReadHandle is used here instead of Handle
            bool bSuccess = NativeMethods.HidD_GetInputReport(ReadHandle, cmdBuffer, cmdBuffer.Length);
            HidDeviceData deviceData = new HidDeviceData(cmdBuffer, bSuccess ? HidDeviceData.ReadStatus.Success : HidDeviceData.ReadStatus.NoDataRead);
            return new HidReport(Capabilities.InputReportByteLength, deviceData);
}
```

closed 2023-07-25 by alidayan