mikeobrien/HidLibrary #90: Unable to get data at client side

ahsanned1, 2017-10-07:
Hi All,

I am trying to read Data from AccessIS MSR (Model : ACR75ABN2). When I written my code directly on a windows Form I am able to read data from MSR device but when I moved that code in a class and trying to call from Windows form I am not able to access the read data but data is in my class's property ReceivedData. Can anyone please suggest what is the issue? I stuck here...any help will be appreciated.

ahsanned1, 2017-10-07:
Below is my Code for the class:
using HidLibrary;
using System;
using System.Collections.Generic;
using System.Globalization;
using System.Linq;
using System.Text;
using System.Threading.Tasks;

namespace MagneticStripReaderUtility
{
    public class MagneticStripReader
    {
        private const int VendorId = 0x0db5;
        private const int ProductId = 0x015b;

        public delegate void ReadHandlerDelegate(HidReport report);
        private List<HidDevice> _lstDevice;
        private HidDevice _device;
        public string _receivedData = string.Empty;
        public bool IsDeviceRemoved = false;
        public DateTime? LastReadHandlerCalledAt = null;
        public bool IsDataReadComplete = false;
        int readCounter = 0;

        public string ReceivedData
        {
            get
            {
                return this._receivedData;
            }
            set
            {
                this._receivedData = value;
            }
        }

        public MagneticStripReader()
        {
            _lstDevice = HidDevices.Enumerate(VendorId, ProductId).ToList();
            _device = _lstDevice[0];
            ReceivedData = string.Empty;
            _receivedData = string.Empty;
        }

        public void OpenDevice()
        {
            if (_device != null)
            {
                _device.OpenDevice();
                _device.Inserted += DeviceAttachedHandler;
                _device.Removed += DeviceRemovedHandler;

                _device.MonitorDeviceEvents = true;
                ReceivedData = string.Empty;
                _receivedData = string.Empty;
            }
            else
            {
                
            }
        }
        public void CloseDevice()
        {
            _device.CloseDevice();
            IsDataReadComplete = true;
            ReceivedData = string.Empty;
            _receivedData = string.Empty;
        }

        public bool StartReading()
        {
            bool rtnReadStarted = false;
            try
            {
                _device.ReadReport(ReadHandler);
                rtnReadStarted = true;
                IsDataReadComplete = false;
            }
            catch (Exception ex)
            {
                rtnReadStarted = false;
            }
            return rtnReadStarted;
        }

        private void ReadProcess(HidReport report)
        {
           //BeginInvoke(new ReadHandlerDelegate(ReadHandler), new object[] { report });
        }

        private void ReadHandler(HidReport report)
        {
            if (report.Data[0] == 0 && report.Data[1] == 0 && report.Data[2] == 0 && report.Data[3] == 0 && report.Data[4] == 0 && readCounter != 0)
            {
                _device.CloseDevice(); 
                _device.Dispose();

                ReceivedData = _receivedData;
                IsDataReadComplete = true;
            }
            else
            {
                readCounter++;
                LastReadHandlerCalledAt = DateTime.Now;
                if (!_device.IsConnected) { return; }

                String str = String.Join(" ", report.Data.Select(d => d.ToString("X2")));
                string[] strArray = str.Split(' ');

                char ch = 'a';
                foreach (string s in strArray)
                {
                    ch = (char)Int16.Parse(s, NumberStyles.AllowHexSpecifier);
                    _receivedData += ch.ToString();
                }
                _receivedData += " ";
                _device.ReadReport(ReadHandler);
            }
        }

        private void DeviceAttachedHandler()
        {
            IsDeviceRemoved = false;
            _device.ReadReport(ReadHandler);
        }

        private void DeviceRemovedHandler()
        {
            IsDeviceRemoved = true;
        }
    }
}

Below is how I am reading card data in a textbox from above class:
namespace MagneticStripReaderUtility
{
    public partial class Form1 : Form
    {

        private System.Timers.Timer timerReadCompleted = null;
        double timerInterval = 2000;
        bool IsReadStarted = false;
        MagneticStripReader msr = null;
        
        public Form1()
        {
            InitializeComponent();
        }

        private void Form1_Load(object sender, EventArgs e)
        {
            msr = new MagneticStripReader();
            msr.OpenDevice();
            IsReadStarted = msr.StartReading();

            if (IsReadStarted)
            {
                timerReadCompleted = new System.Timers.Timer(timerInterval);
                timerReadCompleted.Elapsed += new System.Timers.ElapsedEventHandler(CallBackDataReadCompletedFunction);
                timerReadCompleted.Enabled = true;
                timerReadCompleted.AutoReset = true;
                timerReadCompleted.Start();
            }
        }

        private void CallBackDataReadCompletedFunction(object obj, System.Timers.ElapsedEventArgs e)
        {
            try
            {
                if (msr.IsDataReadComplete)
                {
                    timerReadCompleted.Stop();
                    txtReceived.Invoke(new Action(() => txtReceived.Text = msr.ReceivedData));
                    //txtReceived.Text=msr.ReceivedData;
                }
            }
            catch (Exception ex)
            {
                txtMessage1.Text = ex.InnerException.Message;
            }
        }
    }
}
 

Please let me know where I am wrong.

Thanks

conradsson, 2018-01-09:
I have a similar issue. Trying to read card ID with a OMNIKEY 5427CK and I'm also unsuccessful in this task. 
Maybe you have solved it **ahsanned1**?