mikeobrien/HidLibrary #91: Issue with ReadFile() using Griffin Powermate

jkqd, 2017-10-31:
Hi,

Before I get into my issue, thank you very much for the library.

So, I'm having an issue with the NativeMethods.ReadFile() method (called in HidDevice:ReadData(int timeout) and running not overlapped) in HidDevice.cs. This is on Windows 7 and 10, both 64 bit machines. I'm using this library to connect to the Griffin Powermate.

Issue:
The library and device will be working correctly, and then for some reason it goes awry. When we press the button down on the Powermate, the buffer returned by the ReadFile method returns byte buffers indicating two "Button Up" events instead of the "Button Down" and then "Button Up" events one would expect. And when you rotate the Powermate knob in either direction, the ReadFile method does not return until the knob has been displaced far enough such that two displacement events are returned. And this behavior continues until either the Powermate is unplugged and plugged back in, or sometimes the behavior will randomly stop and the Powermate gets back on track and starts behaving correctly again. This is the typical wonky behavior. Some more rare behaviors when the device is stuck in this mode is that we occasionaly get two "Button Down" events instead of two "Button Up" events. Or random button events show up later in between knob displacement events.

Not sure if this is within the scope of this project, nor if it is a bug in this library. It seems like it might be a problem in the Griffin Powermate hardware/firmware and its interaction with the Windows Kernel32.dll that NativeMethods is wrapping. But, I was wondering if anyone else has seen this issue and knew how to go about fixing it. 

Thanks,
Jeff

pctechtv, 2017-11-26:
Hi Jeff, could you explains more about what this relates to. I too have questions about the Power Mate. Are you creating a driver? I had wondered if there is an SDK for this unit? Could you tell me some more about your project and what you are doing. Thanks

jkqd, 2017-11-29:
Sure.
We were trying to see if we could use the Powermate as a controller to drive some inputs into some hardware. Move left, move right kind of thing. As for a driver, we were using this project, as well as the Powermate example in HidLibrary/Examples from thammer, to convert the output of the Powermate into events that we can capture in our software.  Not sure about an SDK. I think it has an application where you can map the rotations and clicks to various controls, but I haven’t looked into it much. Hope this helps…

pctechtv, 2017-11-29:
Got it!. I am was trying to get better results out of it as a USB device to
control software like Adobe Premiere. Being a programmer I wish they had an
SDK to develope you own driver for it. I could not get it XP Vista driver
to play nice until I got creative with it. Basically, when I would program
commands into it the program (Premiere etc.) was receiving weird stuff and
not the commands. So I did what I do with other controllers. That is let
them send virtual keys (ones not on the standard keyboard e.g. F23) and let
Quick Macros do the talking to the applications. Here are some pics of how
it works

​
​
​