mikeobrien/HidLibrary #102: Getting device capabilities

StainlessTobii, 2018-06-21:
Hi Mike,

Been doing some more work as I am trying to work out what buttons / switches are available on a device and automatically build a mapping table for the incoming bit stream.

So far I have extended your code to include Hidp_GetValueCaps and Hidp_GetButtonCaps. I have also added a few methods to your extension class to map from usage to something readable,

I modified the test harness to display this data, and I am totally and utterly confused.

Using a standard Xbox controller as a test subject I can see that the device has 6 value caps and one button cap. The button cap contains 9 buttons, which is correct.

The value caps contain everything **EXCEPT** the left rigger. 

`Usage GENERIC

Value caps    6

  UsagePage   GENERIC Usage Y                   Size 16 IsRange False
  UsagePage   GENERIC Usage X                   Size 16 IsRange False
  UsagePage   GENERIC Usage RY                  Size 16 IsRange False
  UsagePage   GENERIC Usage RX                  Size 16 IsRange False
  UsagePage   GENERIC Usage Z                   Size 16 IsRange False
  UsagePage   GENERIC Usage HATSWITCH           Size 4 IsRange False
Button caps    1
  UsagePage    BUTTON Min 5 Max 14`

I can see the device has 16 data indices ... so I am missing one.....  but where?

a) Have you any ideas
b) Do you want me to send you a code drop when I get closer?

StainlessTobii, 2018-06-21:
Using a third party app I was able to get a full report from the same device.

I can see the same problem in this data........... WTF!

Item Tag (Value) | Raw Data
-- | --
Usage Page (Generic Desktop) | 05 01
Usage (Game Pad) | 09 05
Collection (Application) | A1 01
Collection (Physical) | A1 00
Usage (X) | 09 30
Usage (Y) | 09 31
Logical Minimum (0) | 15 00
Logical Maximum (-1) | 26 FF FF
Physical Minimum (0) | 35 00
Physical Maximum (-1) | 46 FF FF
Report Count (2) | 95 02
Report Size (16) | 75 10
Input (Data,Var,Abs,NWrp,Lin,Pref,NNul,Bit) | 81 02
End Collection | C0
Collection (Physical) | A1 00
Usage (Rx) | 09 33
Usage (Ry) | 09 34
Logical Minimum (0) | 15 00
Logical Maximum (-1) | 26 FF FF
Physical Minimum (0) | 35 00
Physical Maximum (-1) | 46 FF FF
Report Count (2) | 95 02
Report Size (16) | 75 10
Input (Data,Var,Abs,NWrp,Lin,Pref,NNul,Bit) | 81 02
End Collection | C0
Collection (Physical) | A1 00
Usage (Z) | 09 32
Logical Minimum (0) | 15 00
Logical Maximum (-1) | 26 FF FF
Physical Minimum (0) | 35 00
Physical Maximum (-1) | 46 FF FF
Report Count (1) | 95 01
Report Size (16) | 75 10
Input (Data,Var,Abs,NWrp,Lin,Pref,NNul,Bit) | 81 02
End Collection | C0
Usage Page (Button) | 05 09
Usage Minimum (Button 1) | 19 01
Usage Maximum (Button 10) | 29 0A
Report Count (10) | 95 0A
Report Size (1) | 75 01
Input (Data,Var,Abs,NWrp,Lin,Pref,NNul,Bit) | 81 02
Usage Page (Generic Desktop) | 05 01
Usage (Hat Switch) | 09 39
Logical Minimum (1) | 15 01
Logical Maximum (8) | 25 08
Physical Minimum (0) | 35 00
Physical Maximum (4155) | 46 3B 10
Unit (None) | 66 0E 00
Report Size (4) | 75 04
Report Count (1) | 95 01
Input (Data,Var,Abs,NWrp,Lin,Pref,Null,Bit) | 81 42
Report Size (2) | 75 02
Report Count (1) | 95 01
Input (Cnst,Var,Abs,NWrp,Lin,Pref,NNul,Bit) | 81 03
Report Size (8) | 75 08
Report Count (2) | 95 02
Input (Cnst,Var,Abs,NWrp,Lin,Pref,NNul,Bit) | 81 03
End Collection | C0

StainlessTobii, 2018-06-23:
Solved one issue.... for some unknown reason the left trigger and the right trigger are combined to give the Z value.

This is just madness, but what the hell.

mikeobrien, 2018-06-25:
Yeah, I have no idea, I haven't touched HID in over a decade so wont be much help. :(

StainlessTobii, 2018-06-27:
No problem, I will hack it apart and figure out what is going on.
Then I will post the code somewhere you can grab it if you want.