mikeobrien/HidLibrary #107: I plan to make XML docs for HidLibrary, how would you prefer I submit PRs?

akmadian, 2019-03-26:
I'd be happy to work on some XML docs for HidLibrary, in fact I've already started!

When I submit PRs, would you prefer that I submit a PR for each newly documented class, or just get it done all at once and PR then? I ask because my knowledge of HID communication is ok, but not great, and I want to be sure that all of the docs are factually correct in terms of protocol and implementation. I figure that submitting incremental PRs gives the maintainers a chance to  ok all new docs as they come in, instead of getting a massive dump all at once.

Thanks, I love the project, and it's been a big help to me in my work!

akmadian, 2019-04-09:
Pinging @mikeobrien cause it's been a couple weeks

mikeobrien, 2019-10-01:
Hey thanks for the offer and sorry for the late reply! Life has been crazy this year... Yeah, if you want to submit a PR I'll get it merged. Thx again!