mikeobrien/HidLibrary #112: Data just contains 0's

rob-lutrons, 2019-06-13:
Hi,

First time user of HidLibrary.

I followed the simple example. My OnReport method gets called continuously and report.Data is always and empty byte[8].

My device is a barcode scanner which works as a HID keyboard. When I have notepad open the scanner outputs to USB so I'm assuming it's working.

I should be replacing ReadReport(OnReport) with a call to Read after the scanner has scanned? Is it possible it just isn't supported?

OverchenkoDev, 2019-12-08:
I have the same issue. My report data is always empty byte[3] which is being converted to "AAAA" in string.
Did you find any solution for it?

rob-lutrons, 2019-12-09:
No. I changed my scanner to serial mode.

benedekkupper, 2022-02-02:
This seems like a duplicate of #73