mikeobrien/HidLibrary #123: Trying to read report

Asken, 2020-11-11:
`var devices = HidLibrary.HidDevices.Enumerate(1848);
            device = devices.FirstOrDefault();
            device.OpenDevice();
            //device.MonitorDeviceEvents = true;
            device.ReadReport(ReadReportCallback);`

Gives me `System.PlatformNotSupportedException: 'Operation is not supported on this platform.'`

I'm on win 10 running .Net 5

Asken, 2020-11-11:
I used the source and then it's something else :)

closed 2020-11-11 by Asken

olegtar83, 2021-10-04:
> I used the source and then it's something else :)

Hi, i got the same issue, can you share how you fixed it?

Asken, 2021-10-04:
I hardly remember but I think I used the source from git rather than the nuget package

olegtar83, 2021-10-05:
Nah.. it didn`t worked.. do you rewrite the BeginInvoke() method, could you check?