mikeobrien/HidLibrary #132: Operation is not supported on this platform Errors in .Net Standard 2.1 or Greater

youst03, 2022-01-27:
I believe the following needs to be updated as "NETSTANDARD2_0_OR_GREATER || NET35 " to handle projects of .net standard > 2.0

#if **NET20** || NET35

XySoftNeil, 2022-07-11:
BeginInvoke is the "not supported operation" if you are trying to use .NET 6 - Refactor to replace it with Invoke where needed. I would be willing to do a pull request but there are 3 sitting there and I'm not sure this is a living project.

mmjc23, 2022-08-08:
I get the same error with the _MonitorDeviceEvents_ property and the Core 3.1 framework

radop, 2023-01-05:
I have problem with hidlibrary

i create a Form windows desktop application in c#

_device.MonitorDeviceEvents=true

that command throws exception: PlatformNotSupportedException

i use MS visual studio 2022 c#, and Windows home 11, 64.bit

By console program workt all ok

emaiutto, 2023-03-04:
You need to change the code for this: Here it is working perfectly and faster. The code for this project needs a major cleanup. Replacing "invokes" and removing "delegates" is critical. Using TASK you don't need any of that!

        public async Task<HidReport> ReadReportAsync()
        {
            return await Task.Run(() => ReadReport());
        }

to consume it you just have to do this:

            ReportDevice = Task.Run(async () => await Device?.ReadReportAsync());

            ReportDevice.ContinueWith(t => OnReport(ReportDevice.Result));

I will upload the modifications when they are ready and clean. In my case it works perfectly with a Saitek Radio Panel for my flight simulator. (Read / Features)

greetings to all
Esteban

C00lzer0, 2025-01-21:
Same error in 2025 @ _device.MonitorDeviceEvents = true;
Anything new?