mikeobrien/HidLibrary #134: NuGet package update ?

Mutex666, 2022-02-15:
Hi, is there a possibility that the NuGet package of the library could be updated ?  Looks like it was not updated since the end of 2019 :-\

Or maybe I am missing something.

bj-rn, 2024-02-05:
Hey, since the repo seems to be in hibernation and this issues hasn't received an answer in 2 years I went ahead and created an [unofficial nuget package](https://www.nuget.org/packages/HidLibrary.Unofficial) for the purpose of providing support for newer frameworks. I removed the old targets (_net20_, _net35_, _net40_, _net45_,  _netstandard2_) and added _net48_, _net50_, _net60_, _net80_. I only did very limited testing with _net60_ pertaining to my usecase.
If you decide to "officially" update the nuget or want me to remove the package for other reasons please get in touch via github.

mikeobrien, 2024-02-05:
Hey, sorry, I've been slammed with non dev stuff so haven't had the time to maintain this. I'll try to get the CI/CD pipeline rehabbed and start getting nugets published again. I'll keep you posted.